Dan-in-CA/SIP #310: wrong end time with sequential

Kali-, 2023-09-08:
When sequential is set in options and multiple stations are selected in a single program the duration does not take care of how many stations are selected

Dan-in-CA, 2023-09-09:
Could you provide a bit more detail.
When I run a program with multiple stations in sequential mode the log shows the duration for each station correctly.

What version of sip are you using?
Are you referring to a program with individual durations set for each station?
How are you observing the duration that does not take care of how many stations are selected?

Dan

Kali-, 2023-09-09:
I'm running the latest version (5.0.17) and I'm referring to the "Programs" page

This is one of my programs with 2 stations and sequential enabled.

Run: S01, S02
Starting: 05:30
for 00 hrs 35 mins 00 secs until 06:05

and I should expect to see:

for 01 hrs 10 mins 00 secs until 06:40

Dan-in-CA, 2023-09-09:
Thank you. 
Now I can look for a fix.

Dan

Dan-in-CA, 2023-09-12:
I think it is now fixed in version 5.0.18.

Dan

Kali-, 2023-09-13:
yes, it's fixed

ty
Giulio

closed 2023-09-13 by Kali-