Dan-in-CA/SIP #311: Control for Centre Pivots.

RRIoT, 2023-09-09:
Hi Dan.
I love SIP and what you have produced.
Was wondering if you could help with being able to control a Centre Pivot Irrigator. 
I can provide you with all the information. 
Been doing learning IoT for along time, but would like to to learn to program.
I am very good with Electronics and am currently a Technician fixing Centre Pivot Irrigators in Ashburton, New Zealand.
I hope you may be interested in helping.
Have a great day and Thank you.

Dan-in-CA, 2023-09-10:
Yes. sounds interesting.
Sounds like we could learn a lot from each other.
Dan

RRIoT, 2023-09-18:
Hi Dan.

So sorry for late reply.

What is best way to sent you documents.

The Center Pivots are made by:
Zimmatic: www.lindsay.com <http://www.linsay.com>
Reinke: www.reinke.com
Valley: www.valleyirrigation.com

These are the major brands out of the U.S.

They all use their own systems, but they basically work the same.

They have their own remote systems, that are very costly and this is where
I was wondering if we can work together to make a better system for this
where we can also use lorawan to connect to them instead of costly 4G data.

The Zimmatic Irrigation calculator android app which I ran on PC with
Bluestacks, will show the formulas, but can also provide you a pdf of the
formulas from the University of Nebraska in the U.S.

https://m.apkpure.com/zimmatic-irrigation-calculator/com.agilx.zimmaticcalculator

Look forward to hearing from you Dan.

And Thank you very much.

RRIoT, 2023-09-19:
> Try signing up on the SIP forum:
> https://nosack.com/sipforum/index.php
>
> You can send me a private message from there and we can exchange contact
> information.
>
> Dan

Hi Dan

Thank you so much for your reply

Have joined forum as RRiot

Kind Regards

Craig

Dan-in-CA, 2024-05-23:
Closing issue because it is inactive and not a problem report.

closed 2024-05-23 by Dan-in-CA

RRIoT, 2024-05-23:
Hi Dan.
Was wondering if you still were going to design a plugin, etc to be able to
run Centre Pivots on SIP after sending you the information on centre pivots.
Can PM you. Have a great day.