Dan-in-CA/SIP #312: Feature request - adjustments in logs

ctosae, 2023-09-13:
Is it possible to add the total adjustments % in logs?

https://github.com/Dan-in-CA/SIP/blob/8f5050b53b610725c27763d62482626e85384b22/helpers.py#L356C9-L356C20

(and show it in `/vl` page, compatibly with the old logs where it was not present)

Dan-in-CA, 2023-09-13:
Yes. In the log.
I wanted to be sure that is the value you are asking for.
It should be an easy change.

Dan

ctosae, 2023-09-13:
Yes, it can be interesting to analyze over time whether external plugins have worked correctly

jonathan-marsh, 2023-09-13:
Hi Dan,

 

Related log improvement, if you are delving into the logging part of the code 😉. 

 

The log identifies the station by a single value.  If the program has a name, it uses that, if there isn’t a name it uses the pid.  But program names aren’t unique (e.g. several can default to “Unnamed”).  That makes it hard to tell what program actually ran.  That makes the log harder to analyze.  For example I use the program number to color code the web UI display legend, it would be much easier and more stable to have a pid for each entry, and not have that removed when the program has a name.  So if you are making changes to the log file format, perhaps we could also write the pid in addition to the name if there is one, would be helpful.

 

Happy to work on this, but thought I’d mention as it seems related to adding water level adjustment values.

 

-Jonathan

Dan-in-CA, 2023-09-14:
"Unnamed" was a place holder prior to and during development of the program name feature. I did a search of the code and the only place I found it still being used was in convert_progs.py a utility that converts an older version of the programs data file (programs.json) to the new format (programData.json). I replaced that occurrence with an empty string.

There probably are a number of SIP installations that still have the "Unnamaed" place holder in the program date file.
I will add a function to clear those occurrences. I think the best way to get rid of any remaining ones would be to check for and remove "Unnamed" in the program data when the file is read at startup.  As soon as a program is created or modified the file will be saved without the offending place holder and it will not be present in memory.

I will let you know as soon as the updates are on GitHub. It may be a day or so.

Dan

jonathan-marsh, 2023-09-14:
"Unnamed" isn't that big a problem, since it's just a conversion artifact.
But it just illustrates that program names aren't guaranteed to be unique.
Another example, if the user changes the name, the log and the predictions
can't be correlated.  Color coding is easier on program numbers or id, and
since this is readily available at the time the log is written, that seems
better than attempting a lookup after the fact.

-Jonathan

Dan-in-CA, 2023-09-14:
Ah.
I see what you are getting at.
If you wouldn't mind I would appreciate your work on a fix.

Dan

Dan-in-CA, 2023-09-14:
Giving this a bit more thought;
There is a variable gv.pnames that is a list of all program names.
When a user adds a program name it would be easy to check the list for a duplicate and raise an error at that point.

Dan

jonathan-marsh, 2023-09-15:
I don't think I'd consider a duplicate user-assigned name to rise to the
status of error.  As long as the program doesn't rely on unique values the
user can put anything they want in there, including leaving them all
blank.  The name is simply there for UI purposes.

jonathan-marsh, 2023-09-15:
Pushed a fix, plus a couple of other bug fixes/improvements.  I'm still
struggling a bit to get comfortable with git, if you have any advice on how
I should create nice clean granular pull requests I'm happy to learn!

-Jonathan

Dan-in-CA, 2023-09-15:
@ ctosae,
SIP version 5.0.20 now has the adjustment added to the log.
Dan

activenodes, 2023-09-15:
> @ ctosae, SIP version 5.0.20 now has the adjustment added to the log. Dan

Thanks, just updated to the last version (5.0.22).
I noticed that it's not possible to scroll the log page "left<>right" from mobile (iOS).

activenodes, 2023-09-15:
Sorry @Dan-in-CA and @jonathan-marsh but the issue with the scroll is in all sections (homepage too).
It is no longer possible to use SIP from mobile

ctosae, 2023-09-15:
I just updated to the latest version (5.0.22).
I noticed that it is no longer possible to scroll "left<>right" from mobile (iOS) in all sections (homepage included).
So with this version it's no longer possible to use SIP from mobile.
Executed a rollback

jonathan-marsh, 2023-09-16:
Sorry about that, dumb mistake to change CSS with larger scope than needed
to fix my display issue.  Generating a pull request now.

ctosae, 2023-09-16:
@jonathan-marsh CSS fix is working, thanks!

@Dan-in-CA just a question... (pre-upgrade) it asked me to update from 5.0.18 to 5.0.48

[image]

later it show me that I'm up-to-date with the 5.0.24

[image]

Dan-in-CA, 2023-09-16:
The version number is updated automatically and involves a calculation using the git rev-list --count.
There was a brief time when the count was out of sync. That is what you saw.
Dan

ctosae, 2023-09-16:
@Dan-in-CA a little consideration.
should probably be denoted as "---" in case of Run-once.
In that case the duration is not affected by adjustments

Dan-in-CA, 2023-09-16:
I pushed a update that shows "---" adjustment for run once, manual, and node-red programs.
Dan

ctosae, 2023-09-16:
Super! 🥇

closed 2023-09-18 by ctosae