Dan-in-CA/SIP #316: Home  "<< PREV DAY" is not working in Firefox (PC)

ctosae, 2023-09-16:
With Firefox (PC) is not possible to go back before today.

- Clicking `<< PREV DAY` the date is still the same (current).

- Clicking `NEXT DAY >>` the date is still the same (current) until the number of clicks to "PREV DAY". After this, the date goes foward, but using `<<PREV DAY` it's never possible to go back before today.

Working correctly with iOS (Mobile) and Chrome (PC)

Dan-in-CA, 2023-09-16:
I am seeing something similar.
On both firefox and chrome PC the << PREV DAY works but the NEXT DAY >> does not.
Dan

Dan-in-CA, 2023-09-16:
In Firefox JavaScript console clicking the next day button raises the error:
Uncaught TypeError: progs[(p - 1)] is undefined
the error seems to be in 
schedule.js
function programName(p)

The problem goes away if all programs have a name.

Dan

ctosae, 2023-09-16:
AFK but all of my programs have a name and the issue for me is only for back.
Maybe I used "run once" in the past days. Not sure if can be a similar problem with names.
I think it appeared with SIP version 5

jonathan-marsh, 2023-09-16:
I suspect the browser issue is a cache problem - ctrl-refresh should show
the problem in all browsers.

The function doesn't check that p is valid (<0 or >number of programs)  The
latter could occur if programs weren't named, a high program number was
logged, and the program subsequently deleted.  We could put in a check for
this.  Or just put a try/catch around it so any failure to find the name
isn't fatal.

Dan-in-CA, 2023-09-16:
Adding a "console.log("p: ", p)" line in function programName(p)
shows that when the error occurs:  p:  <empty string>
Dan

ctosae, 2023-09-16:
I already tried CTRL+F5 but in my case it didn't solve the problem

If I remember correctly the problem appeared after the upgrade from version 4 to 5 but I only realized it now

Dan-in-CA, 2023-09-17:
This morning it seems to be working in Firefox and chrome on Windows 11 but not on Firefox or chrome on android phone.
Only the next day button seems to be effected.

I also tested earlier SIP revs on windows and didn't see the problem.

Very strange.

Dan

Dan-in-CA, 2023-09-17:
After clearing the cache the next day button stopped working in Firefox and chrome on Windows.

Dan

Dan-in-CA, 2023-09-17:
Getting close.
The problem first appeared in rev 5.0.22 in schedule.js
Looking for the exact problem.

Dan

ctosae, 2023-09-17:
Tested again with "anonymous browsing" (like without cache). In my case the problem is as first post.
I seem to remember that it appeared after upgrading from v4 to v5 but I could be wrong

Dan-in-CA, 2023-09-17:
The cause was a change in schedule.js line 109 from
program: pid_array[sid], // program number
to
program: progs[pid_array[sid]-1].name, // program name
Now it is a matter of finding a fix.

Dan

ctosae, 2023-09-18:
💯

closed 2023-09-18 by ctosae