Dan-in-CA/SIP #318: No planning after midnight

ctosae, 2023-09-18:
Regarding this https://github.com/Dan-in-CA/SIP/commit/8da1d7dbe59b026b98630459d2ae67f49fbf6a0e I still have a similar problem.

In my case [Local version 5.0.30 (2023-09-17)]

With this example: **2 Programs**
- A) starting at: 03:30
- B) starting at: 23:00

Local time: GMT+2

At midnight the planning shows no schedule.
In the morning, the 3.30 program (A) was executed and in the planning I also see the evening one (B)

Maybe @jonathan-marsh can help us

jonathan-marsh, 2023-09-18:
> Regarding this 8da1d7d

Happy to look into it.  I'm not yet successful in replicating the issue,
even though I have a set of programs starting at 23:00 every night and
running for 5 hours.

Can you provide more information about the situation in which the display
is incomplete?  It sounds like the future schedule is ok but the history
(based on the log) is not displaying correctly.  Can you share the log
entry for that run?

-Jonathan

ctosae, 2023-09-19:
Thank you!
I'm not talking about the log but about real-time planning.
At midnight any schedule for the current day disappears...to reappear later.
My setup:

```
# ls -l /etc/localtime
lrwxrwxrwx 1 root root 31 Mar  3  2023 /etc/localtime -> /usr/share/zoneinfo/Europe/Rome
```

**sd.json**
```
{
    "alr": 0,
    "bsy": 0,
    "en": 1,
    "htip": "192.168.1.111",
    "htp": 8888,
    "idd": 0,
    "ir": [
        0
    ],
    "iw": [
        0
    ],
    "lang": "default",
    "lg": 1,
    "loc": "",
    "lr": 0,
    "mas": 0,
    "mm": 0,
    "mo": [
        0
    ],
    "month": 9,
    "mtoff": 0,
    "mton": 0,
    "name": "Home",
    "nbrd": 1,
    "nopts": 13,
    "nst": 8,
    "passphrase": "xxxxxxxxxxxxxx",
    "pigpio": 0,
    "rbt": 0,
    "rd": 0,
    "rdst": 0,
    "rs": 0,
    "rsn": 0,
    "rst": 1,
    "sdt": 0,
    "seq": 1,
    "show": [
        7
    ],
    "snlen": 32,
    "tf": 1,
    "theme": "basic",
    "tu": "C",
    "tz": 48,
    "upas": 0,
    "urs": 0,
    "wl": 100,
    "wl_monthly_adj": 100,
}
```

**programData.json**
```
[
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            2700
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "A",
        "start_min": 210,
        "station_mask": [
            1
        ],
        "stop_min": 255,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            1950
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "B",
        "start_min": 1380,
        "station_mask": [
            6
        ],
        "stop_min": 1412.5,
        "type": "interval"
    }
]
```

ctosae, 2023-09-20:
Tested this:
```
# timedatectl set-ntp false
# timedatectl set-time '2023-09-20 23:59'

... 2 min later ...

# date
  Thu Sep 21 12:01:07 AM CEST 2023
```
But the web time is always the current

[image]
It might be something with this and not with the system

I'll try changing the PC time too

Dan-in-CA, 2023-09-20:
The time displayed on the UI is supposed to be from the time on the device running SIP.
If someone is accessing SIP from a remote location in a different time zone they should see the time at the SIP location.

Dan

jonathan-marsh, 2023-09-20:
> Thanks @Dan-in-CA <https://github.com/Dan-in-CA>
> I was just trying to simulate this: #318 (comment)

I'm trying to duplicate this, and find that if you are viewing the schedule
as midnight ticks over, that the display doesn't update its concept of
"today."  The date shown is the one that just ended, not the next day. If
you click on "Today" you advance to the new day.  If you don't advance the
day, you won't see any projected activity, only log entries.  Even log
entries can be a bit off - their display can be delayed until the program
ends (when the log entry is created) and even then might not appear until
the page is the page is refreshed (the display assumes the log is immutable
history and doesn't poll for updates).

Does one or more of these issues seem like the problem you are seeing or is
there something else going on?

-Jonathan

ctosae, 2023-09-21:
Here is a real situation (this night).. I was with mobile

**20 September 23:57**
- Watering scheduled for tomorrow (21 September 03:30 + 23:00).
- All ok!
[image]

**21 September 00:01**
- Scheduled irrigations has disappeared.
- This is wrong.
[image]

- It was postponed to the following day (22 September).
- This is incorrec because "interval_base_day" is 3.
[image]

**21 September 04:05**
- Scheduled irrigations is back!
[image]

jonathan-marsh, 2023-09-21:
Thanks for showing me that in detail, I'll try again to reproduce it.

-Jonathan

ctosae, 2023-09-21:
Thank you Jonathan!

Am I the only one who noticed this?

jonathan-marsh, 2023-09-21:
Yes, you seem to be unique with this problem, either you have some special
combination of circumstances or you're the only one up in the middle of the
night to notice it!

I still have not been able to reproduce this successfully, but it sure
looks like a manifestation of the issue I fixed recently, where interval
programs were incorrectly influenced by the time the schedule was viewed.
Can you double-check for me that we are running the same fix?  schedule.js
line 14 should read:
     simday = ...
If simday is defined a few lines further up, outside the interval program
check, you don't have the fix.

I also observed a couple of other things:

1) Showing the display in the device time zone seems to have been disabled:
Line 104 of base.html sets tzDiff to 0, the proper calculation is commented
out.  Dan, any reason for this?  I re-enabled it to assist with my testing
of this issue as I put the software in various timezones, and haven't seen
any untoward behavior.

2) The calendar view requests updates from the device once a minute, if you
are viewing tomorrow's schedule there won't be regular checks.  When the
clock flips over past midnight (so tomorrow becomes today), there isn't an
automatic mechanism for kicking off a refresh and starting the regular
updates.  You have to refresh or click on Today for it to recognize it
needs to start regular updates.  I can put in a fix for this, though
honestly the impact of this on the UI is miniscule.  But I doubt that's the
issue you are having, which sounds like it's unaffected by refreshing the
display.

-Jonathan

Dan-in-CA, 2023-09-21:
@ Jonathon,
Re issue #1. I don't recall disabling that.
Might have happened during some testing a long time ago. 

Dan

ctosae, 2023-09-21:
Thank you guys for the support!

> Yes ...or you're the only one up in the middle of the night to notice it!

ehehe... I noticed it accidentally the first time and didn't understand what was happening

>Can you double-check for me that we are running the same fix?  schedule.js
>line 14 should read:
>     simday = ...
>If simday is defined a few lines further up, outside the interval program
>check, you don't have the fix.

Here my schedule.js
```
// Global vars
var displayScheduleDate = new Date(Date.now() + tzDiff);
var displayScheduleTimeout;
var sid,sn,t;
if (typeof progs !== 'undefined'){var nprogs = progs.length}; // number of programs
if (typeof nbrd !== 'undefined'){var nst = nbrd*8}; // number of stations

function scheduledThisDate(pd,simminutes,simdate) { // check if progrm is scheduled for this date (displayScheduleDate) called from doSimulation
  // simminutes is minute count generated in doSimulation()
  // simdate is a JavaScript date object
  var wd,dn;
  if(pd['enabled']==0)  return 0; // program not enabled, do not match
  if(pd['type'] == 'interval') { // if interval program...
        simday = Math.floor(new Date(simdate.getUTCFullYear(), simdate.getUTCMonth(), simdate.getUTCDate(),Math.floor(simminutes/60), simminutes%60)/(1000*60*60*24))
        if(((simday)%pd['interval_base_day'])!=pd['day_mask']) return 0; // remainder checking
  } else { // Not interval
    wd=(simdate.getDay()+6)%7; // getDay assumes sunday is 0, converts to Monday is 0 (weekday index)
[..]
```

Well.... I'll try to find a way to simulate without waiting for midnight.
Perhaps simply with a Windows PC, Timezone Rome and setting the time to 11.59pm

jonathan-marsh, 2023-09-22:
I've been running various logging exercises around the clock to try to find
out what's going on.  I think there is a fundamental design flaw, in that
interval calculations use UTC, while the device and the UI are generally
using device time.  When the UI tries to calculate the interval by
converting device time back to UTC, the answer it gets may change during
the 24-hour period it's running the simulation over, because the display
time (device time) and UTC time are offset.  Thus projections on program
runs might go forward or backward a day in a fairly unpredictable manner.
I think this present scenario is one such example; it's probably surprising
that it hasn't manifested more often, but it's likely a middle-of-the-night
issue for most users.

Conceptually the solution is pretty simple - make sure we solely use device
time in interval calculations, both in the timing loop and in the
simulations.  It's possible that installing this fix may cause some
interval programs running in the gap to skip forward or backward a day when
the fix gets deployed.

I'm developing and testing such a fix now.  If you have any advice or see
potential side effects let me know.

-Jonathan

bkoblenz, 2023-09-22:
I took a slightly different approach to this kind of problem.

I introduced the notion of "virtual day start" which is the start hour of a
day.
In general, all programs must operate within a virtual day and I generate
warnings for things that do not.  ie, a recurring program must have all of
its recurrences within a virtual day.

For folks who want their watering to (for example) run from 10PM - 8AM
might start their virtual day at 9AM (or 10PM for that matter).

The displays then present as 24 hours starting on a virtual day boundary
and most of the rest of the system is unaffected.

It is a pretty unusual requirement for something to need to cross a virtual
day boundary, but if it exists, I do log an entry at the boundary so
historical records are accurately captured and always displayed.

-brian

jonathan-marsh, 2023-10-05:
After wrestling with timezone issues, I think I've at last got a solid
solution to this.  I've been testing in lots of timezones with DST and
without and looks like the display and the timing loop are fully in sync.

Generating a pull request now.  However, Dan please check the diff on the
schedule.js programName() function - I didn't touch this function but it
looks like my source is out of sync with a change you made even though
github isn't flagging it?  Might want to keep the original with the
node-red test.

-Jonathan

jonathan-marsh, 2023-12-14:
Fixed in #320

ctosae, 2023-12-15:
Thank you guys!
My scheduler is on standby (cold winter here) but let me try it, I'll update you

ctosae, 2023-12-15:
Something is wrong... the schedules are completely different when viewed from PC and Mobile (iOS)

Here 3 days in sequence:

[image]

[image]

[image]

And the same from mobile

[image]

[image]

[image]

ctosae, 2023-12-15:
```
Today: Friday

Sat: PC "Casa-A" 23:00
     iOS "Casa-B" 23:00

Sun: PC "Casa-B" 23:00
     iOS "Siepe" 03:30

Mon: PC "Siepe" 03:30
     iOS "Casa-A" 23:00
```
As you can see they are offset by 1 day, which is the same problem I noticed from midnight until morning

jonathan-marsh, 2023-12-15:
It seems quite difficult for the same code to give different results
depending on the device - the first thing to check is that the code
really is the same!  On the PC you can hold down the control key while
refreshing to make sure the browser has fetched all the latest code
(specifically schedule.js which is a static file and gets cached locally).
I'm not sure how to do that on iPhone other than clearing the complete
cache for all web sites (settings/safari/clear history.)  But I did find
this hack: Turn on AirPlane mode, refresh the page (to fail), then turn
AirPlane mode back off, then refresh the page to get the updated version.
Can you try that?

-Jonathan

ctosae, 2024-05-22:
I have turned my system back on now for the summer that is coming,

upgrading from 5.0.63 (2023-12-06)
to the last version: 5.1.43 (2024-05-11)

I will update this issue shortly ;)

ctosae, 2024-06-06:
Well.. now the problem is more complicated and critical, also with PC.

There was no planning for last night... but yesterday (5 June):

[image]

[image]

Now, programs show the next start is in 2 days (see here)

[image]

but the planning shows it for tonight (6 June)

[image]

@jonathan-marsh @Dan-in-CA please help.. thank you!

ctosae, 2024-06-06:
Yes @jonathan-marsh ,
the problem is present with all browsers (desktop and mobile). I just tried with Edge, which I've never opened
Discrepancy (from desktop to mobile) seems to have disappeared but the problem is definitely evident.

Here the settings, let me know if you need anything else. Thanks!
```
# ls -l /etc/localtime
lrwxrwxrwx 1 root root 31 May 22 09:21 /etc/localtime -> /usr/share/zoneinfo/Europe/Rome
```
programData.json:

```
[
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            2700
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Siepe",
        "start_min": 210,
        "station_mask": [
            1
        ],
        "stop_min": 255,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            1950
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Spruzzatori",
        "start_min": 1380,
        "station_mask": [
            6
        ],
        "stop_min": 1412.5,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 2,
        "duration_sec": [
            300
        ],
        "enabled": 0,
        "interval_base_day": 3,
        "name": "Spruzzatori Casa-A",
        "start_min": 1380,
        "station_mask": [
            4
        ],
        "stop_min": 1385,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 0,
        "duration_sec": [
            300
        ],
        "enabled": 0,
        "interval_base_day": 3,
        "name": "Spruzzatori Casa-B",
        "start_min": 1380,
        "station_mask": [
            4
        ],
        "stop_min": 1385,
        "type": "interval"
    }
]
```

ctosae, 2024-06-07:
This last night/tonight update
As indicated, the visual schedule differed from reality.... 

-  **Shown as scheduled but not running**

[image]

[image]

-  **And in the end the schedule was not executed and disappeared from the plan**

[image]

[image]

ctosae, 2024-06-07:
I reproduced the problem in a fresh install, simply by copying my "data/programData.json"

```
[
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            2700
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Siepe",
        "start_min": 210,
        "station_mask": [
            1
        ],
        "stop_min": 255,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 1,
        "duration_sec": [
            1950
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Spruzzatori",
        "start_min": 1380,
        "station_mask": [
            6
        ],
        "stop_min": 1412.5,
        "type": "interval"
    }
]
```

By creating equal schedules this is the configuration:

```
[
    {
        "cycle_min": 0,
        "day_mask": 2,
        "duration_sec": [
            2700
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Siepe",
        "start_min": 210,
        "station_mask": [
            1
        ],
        "stop_min": 2910,
        "type": "interval"
    },
    {
        "cycle_min": 0,
        "day_mask": 2,
        "duration_sec": [
            1950
        ],
        "enabled": 1,
        "interval_base_day": 3,
        "name": "Spruzzatori",
        "start_min": 1380,
        "station_mask": [
            6
        ],
        "stop_min": 3330,
        "type": "interval"
    }
]

```

And it looks different in code and plan....

```
@@ -1,7 +1,7 @@
 [
     {
         "cycle_min": 0,
-        "day_mask": 1,
+        "day_mask": 2,
         "duration_sec": [
             2700
         ],
@@ -12,12 +12,12 @@
         "station_mask": [
             1
         ],
-        "stop_min": 255,
+        "stop_min": 2910,
         "type": "interval"
     },
     {
         "cycle_min": 0,
-        "day_mask": 1,
+        "day_mask": 2,
         "duration_sec": [
             1950
         ],
@@ -28,7 +28,8 @@
         "station_mask": [
             6
         ],
-        "stop_min": 1412.5,
+        "stop_min": 3330,
         "type": "interval"
     }
 ]

```

How is it possible. Which is the correct one?

jonathan-marsh, 2024-06-25:
Hi, I have been trying to replicate for a couple of weeks without any success.  Unfortunately my tests have been a bit sporadic and I haven’t been able to leave a system running for extended periods in your time zone.

 

However I did find one bug related to interval programs, that was causing the program description to be incorrect.  For instance it might say “starting today” when it should say “starting tomorrow.”  I’ve checked that fix and some other minor cleanup into my fork at https://github.com/jonathan-marsh/SIP.  I’m not completely confident that this would cause the symptoms you have encountered, it is entirely possible?

 

If you do have any free time, perhaps you could see if the problem persists in that version?

 

-Jonathan

jonathan-marsh, 2024-07-19:
I purchased another RPi for testing purposes, installed clean SIP (as of commit #340) and copied in your sd.json and programData.json, only changing the IP address.  I set the RPi time to Rome, and have been checking in on the UI periodically for many days, but still haven't observed the problem :(.  For me the schedule view and the actual runs match up their interval times precisely.  I have watched it tick over midnight a few times (easy as its late afternoon for me) with no problem.  Frustrating!

The presumed cause of the error is that the RPi's python days-since-epoch value used for triggering actual station runs does not precisely match the javascript days-since-epoch value used by the simulation.  This original issue arose from one calculation assuming GMT and the other assuming local time zone, so the simulation was incorrect for a few hours (depending on the time zone) during the day.  The fix I developed attempted to perfectly align these, but it's possible they don't.  The latest small fix was to ensure that even when they were initially aligned, they wouldn't drift over time.

I have created a couple of testing tools for easy monitoring of these core values, if your system still manifests this behavior and you are able to run these tools perhaps this will help me see where too look in the absence of direct experience of the problem.

First, add these lines at 22 of schedule.js, right after the definition of "simday":
		// Display simday for testing purposes
		if ($("#simday").length == 0) {
			$("#status").append("<div id='simday'></div>");
		}
		$("#simday").text(simday);
		//
This will display the days-since-epoch the simulation will use, right under today's date, every time the simulation runs.  The value will change as you go forward in time, but not backward (as the display then is showing log info, not simulation.)  Days-since-epoch is the core value against determining which interval programs run.

Next, I've developed a test plugin to show the days-since-epoch used by the timing loop.  Add these to a file in the plug-ins folder (I named mine dse_report.py but the name doesn't matter.  This will display the current dse value used in the timing loop in the footer.  This value and the value under the date should always, from midnight to midnight, match for "TODAY".  If they don't, incorrect interval simulations could go awry.  I haven't seen any divergence on my system, but if you experience the issue and these values ARE still aligned, that would be a good indication I've been barking up the wrong tree.

    # !/usr/bin/env python
    # -*- coding: utf-8 -*-
    
    # standard library imports
    import json  # for working with data file
    from threading import Thread
    from time import sleep
    
    # local module imports
    from blinker import signal
    import gv  # Get access to SIP's settings
    from sip import template_render  #  Needed for working with web.py templates
    from urls import urls  # Get access to SIP's URLs
    import web  # web.py framework
    from webpages import ProtectedPage  # Needed for security
    from webpages import showInFooter # Enable plugin to display readings in UI footer
    from webpages import showOnTimeline # Enable plugin to display station data on timeline
    
    
    example1 = showInFooter()  #  instantiate class to enable data in footer
    example1.label = u"Days since linux epoch"
    example1.val = 0
    example1.unit = u""
    
    def data_test():
            while True:
                example1.val = gv.dse          
                sleep(1)        
    
    # Run data_test() in baskground thread
    ft = Thread(target = data_test)
    ft.daemon = True
    ft.start()
    
    ### End data display examples ###
    #################################

ctosae, 2024-07-19:
Ehy @jonathan-marsh, first of all I would like to thank you for your interest and commitment.

The HTML code of the `schedule.js` is probably is not formatted correctly (any HTML code has been removed and I'm getting errors). Could you put it in a "code" section?

I have also prepared a clean virtual environment (which you can connect to and experiment with), the problem is always present here. I will try to contact you via Linkedin to send you credentials

jonathan-marsh, 2024-07-19:
Oops, sure:

`		// Display simday for testing purposes
		if ($("#simday").length == 0) {
			$("#status").append("<div id='simday'></div>");
		}
		$("#simday").text(simday);
		//`

jonathan-marsh, 2024-07-22:
> @jonathan-marsh <https://github.com/jonathan-marsh> here the result:
>
> image.png (view on web)

Excellent, very helpful in locating the issue to the Javascript side.  I'm
running some new tests.

jonathan-marsh, 2024-08-19:
While I still wasn't able to observe the issue firsthand, I was able to write a plug-in that tested the UI values against the Python values for days-since-epoch (using the enhanced plugin architecture).  To make these values match precisely I recoded the Javascript function to directly parallel the Python one.  It appears (at least from my test plugin) that this fixes the issue.  The new schedule.js is at https://github.com/jonathan-marsh/SIP/blob/master/static/scripts/schedule.js.   Since this problem has been so elusive, it would be great to know the fix works for you before committing it to the main branch.  Let me know if you are able to give it a whirl.

ctosae, 2024-08-20:
Thanks @jonathan-marsh for your time ❤️ 
Looks like you solved the problem! 🥇 
Give me a few days and I'll let you know the final result

ctosae, 2024-08-22:
The final result is: **it works**! 👍 
@jonathan-marsh could you commit it to the main branch?
Thanks again!

ctosae, 2024-08-26:
Fixed with:

```
Local SIP version: 	5.1.76
Local SIP date: 	2024-08-24 
```

Thanks!

closed 2024-08-26 by ctosae