Dan-in-CA/SIP #319: Station running (00:00)

ctosae, 2023-09-24:
This morning I found this state: 

[image]

It was 9:30am and the scheduling was for 3:30am
I had to push the button “`STOP ALL STATIONS`”

As you can see my plugins set "Total adjustment 0%", 
so the **station should not start**...

@Dan-in-CA could it depend on this https://github.com/Dan-in-CA/SIP/issues/309#issuecomment-1715843721 ?
_a bug that would raise an error if a duration for a program was less than 30 seconds._

Thanks

Dan-in-CA, 2023-09-24:
I did a quick test with a plugin adjustment set to 0% and the stations did not start.
However the home page did show the station scheduled to run for (00.00)  but not running.

That is a bug.

Dan

ctosae, 2023-09-24:
Yes I can confirm that is not running, only web page like your description.
It has never happened before... often my plugins set to 0%.
I think it's introduced with that change

ctosae, 2023-09-25:
Thank you! I will try it

Dan-in-CA, 2023-09-26:
An unexpected benefit of the new node-red SIP extension helped find the problem.
https://en.wikipedia.org/wiki/Node-RED

A simple node-red flow can read SIP variables while the program is running.
[image]
It showed that in the gv.rs (run schedule) var station 2 had a station number but not a duration other  value.

ctosae, 2023-09-27:
Working fix! Thanks @Dan-in-CA 

**Node-Red**: I have heard high praises from those primarily engaged in developing IoT devices.
The implementation in SIP is excellent; I'm extremely pleased with this project 🥇 

It’s a pity that our summer and irrigation period only lasts from May to the end of September approximately.
Soon, it will have to hibernate for a while...

closed 2023-09-27 by ctosae