Dan-in-CA/SIP #323: wrong adjustment logged

Kali-, 2023-10-13:
Not a big issue but I just noticed that
I have 4 valves that run in sequential on a single program and timer is set to 32 minutes
As you can see in the log, the last one shows a different Adjustment but same duration (maybe the duration is computed at start of the program and adjustment at the end of each valve?)

Date | Start Time | Station | Duration | Adjustment | Program
-- | -- | -- | -- | -- | --
2023-10-12 | 07:36:09 | Prato Retro Parco | 17:03 | 48 % | 3 volte a settimana
2023-10-12 | 07:19:06 | Prato Retro casa | 17:03 | 55 % | 3 volte a settimana
2023-10-12 | 07:02:03 | Prato Fronte Casa | 17:03 | 55 % | 3 volte a settimana
2023-10-12 | 06:45:00 | Prato Fronte Parco | 17:03 | 55 % | 3 volte a settimana

Dan-in-CA, 2023-10-17:
I am looking into the problem.

Dan-in-CA, 2023-10-17:
I set up a test program but did not see the problem.
[image]

@Giulio,
do you have another program that runs after the one that shows the problem?

Kali-, 2023-10-19:
I have only this program enabled and yes I use Weather-based Water Level and monthly adjustment
no station ignores the plugin adjustments

maybe the "weather based ..." do some computation differently for some reason

and this "cosmetic" bug happened to me 3 times 

Date | Start Time | Station | Duration | Adjustment | Program
-- | -- | -- | -- | -- | --
2023-09-29 | 06:45:21 | Prato Retro Parco | 25:07 | 80 % | bi weekly
2023-09-29 | 06:20:14 | Prato Retro casa | 25:07 | 80 % | bi weekly
2023-09-29 | 05:55:07 | Prato Fronte Casa | 25:07 | 80 % | bi weekly
2023-09-29 | 05:30:00 | Prato Fronte Parco | 25:07 | 78 % | bi weekly

Dan-in-CA, 2023-10-19:
The duration is calculated from the number of seconds that a station runs.
32m = 1920s
25:07 = 1507s
1507 / 1920 = 0.784... or 78%

It may be that the weather plugin is changing the total adjustment after the program starts. This would not change the run time of the stations that are already scheduled but would change the adjustment value shown in the log.

The adjustment shown in the log are added per station when the log is updated. This is necessary if a station is set to ignore plugin adjustments on the Stations page.

[image]

[image]

Kali-, 2023-10-20:
Thanks for the clarification (adjustment % can contain decimals but only the integer part is logged)
I also got your point on why the adjustment is added at the end, the only issue is that it can cause confusion (this is why we are discussing here :) )
Monthly adjustment in September was 80 and 55 in october

Dan-in-CA, 2023-10-20:
The adjustment value shown in the log is rounded up to the nearest integer but it is for informational purposes and is not used to control anything. It may include a small amount of "rounding error"

As it is now the log shows the adjustment value from a variable that holds the total adjustments from plugins as well as the water level setting on the home page.  It will probably be better to calculate the log adjustment value from (final run duration / duration set in the program). This should give a more accurate value and will not be affected by adjustment changes that occur after the program starts.

I will work on the change..

Dan-in-CA, 2023-10-21:
SIP rev 5.0.41 has the update.

closed 2023-11-21 by Kali-