Dan-in-CA/SIP #334: git cannot run when SIP directory has non-root ownership

noisymime, 2024-05-08:
### PROBLEM

I recently updated my SIP installation (Including OS level) and noticed that the revision and revision date in the bottom right corner were not being updated correctly. Checking through the logs this was due to git not being able to run on the install directory. Recent versions of git (~2022 onwards, including Raspbian buster-security and up) have been updated to patch CVE-2022-24765 which prevents git from being run on a directory with a difference ownership to the user running git. 

In my case the install directory was owned by my user ID but the systemd service was being run by root and hence the following appears in the log:

```
fatal: detected dubious ownership in repository at '/home/pi/OSPi'
To add an exception for this directory, call:
git config --global --add safe.directory /home/pi/OSPi
Could not use git to determine version! Command '['git', 'rev-list', '--count', 'HEAD']' returned non-zero exit status 128.
```

A similar error is shown for the date of last commit as well. 

If the install script is used I don't believe this will be a problem as the install directory will be owned by root, but if the user has manually installed SIP with their own ownership, then this issue will be experienced. 

### FIX

I'd suggest that a line similar to the below be added to the bottom of `sip_service.sh`:

`sudo git config --system --add safe.directory ${install_location}`

This will bypass this check for the specific install directory and is not a security concern.

Dan-in-CA, 2024-05-08:
Thanks for the input on this and the suggested fix..

Dan-in-CA, 2024-05-11:
I used your suggestion and added the line to sip_service.sh.
Also added some code to gv.py to enable git safe.directory in the core program.

This issue also affected the system update plugin.

closed 2024-05-11 by Dan-in-CA