Dan-in-CA/SIP #339: Interval program works fine, but timeline is wrong

Kali-, 2024-06-19:
i have set 2 programs with 2 day interval one start at 0 and one start at 1 and they work as expected

but looking the timeline in the homepage it wrong as the order is reversed

ctosae, 2024-06-19:
Could you check if it's the same: https://github.com/Dan-in-CA/SIP/issues/318 ?

This is the last update of the issue: https://github.com/Dan-in-CA/SIP/issues/318#issuecomment-2154341544

Kali-, 2024-06-21:
not sure is the same issue, i have 2 program and they run alternately every day

this i my 2 programs:
[image]
[image]

this is the timeline
yesterday
[image]

today
[image]

tomorrow
[image]

the day after tomorrow
[image]

as shown, the timeline of tomorrow says it will run the same program as today, but tomorrow SIP will run the other one
so in the past the timeline is correct, in the future they are inverted

jonathan-marsh, 2024-07-08:
While I haven't been able to reproduce this exactly, I just created a pull request for a related bug that could be causing this as well.  After the request gets accepted, please update and retest and if it still occurs send me additional information - e.g. timezone of the system.

ctosae, 2024-07-09:
Thank you @jonathan-marsh for your support.
I tested it manually, but in my case it didn't solve the problem.
This is the local time: `/etc/localtime -> /usr/share/zoneinfo/Europe/Rome`
Note. I'm using it with docker.. can you try this setup? It's easy to reproduce with all steps here #338 
When it is accepted we await feedback from @Kali- as well

Kali-, 2024-07-15:
no, still not working properly
also for the 1st time the program crashed

Jul 15 14:00:21 orangepipc2 sip[548924]: Traceback (most recent call last):
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "/opt/sip/web/application.py", line 288, in process
Jul 15 14:00:21 orangepipc2 sip[548924]:     return p(lambda: process(processors))
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "/opt/sip/web/application.py", line 699, in processor
Jul 15 14:00:21 orangepipc2 sip[548924]:     h()
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "/opt/sip/web/application.py", line 805, in __call__
Jul 15 14:00:21 orangepipc2 sip[548924]:     self.check(mod)
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "/opt/sip/web/application.py", line 826, in check
Jul 15 14:00:21 orangepipc2 sip[548924]:     reload(mod)
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "/usr/lib/python3.10/importlib/__init__.py", line 169, in reload
Jul 15 14:00:21 orangepipc2 sip[548924]:     _bootstrap._exec(spec, module)
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "<frozen importlib._bootstrap>", line 601, in _exec
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "<frozen importlib._bootstrap>", line 174, in __exit__
Jul 15 14:00:21 orangepipc2 sip[548924]:   File "<frozen importlib._bootstrap>", line 126, in release
Jul 15 14:00:21 orangepipc2 sip[548924]: AttributeError: 'NoneType' object has no attribute 'get_ident'
Jul 15 14:00:21 orangepipc2 sip[548924]: ::ffff:94.156.10.163:57058 - - [15/Jul/2024 14:00:21] "HTTP/1.1 POST /goform/set_LimitClient_cfg" - 500 Internal Server Error

ctosae, 2024-08-26:
@Kali- problem solved. Try updating to:
```
SIP version: 	5.1.76
SIP date: 	2024-08-24 
```

Kali-, 2024-09-03:
confirmed!

closed 2024-09-03 by Kali-