Dan-in-CA/SIP #341: ACKNOWLEDGE import not found on windows

jonathan-marsh, 2024-07-25:
Commit 5ee7f92 introduced this line:

18 +   from multiprocessing.reduction import ACKNOWLEDGE

On windows (my dev environment) this fails, as ACKNOLWEDGE only seems to be defined in reduction.py for Linux systems, and can't be successfully imported on Windows systems.  I.e. ACKNOWLEDGE doesn't seem to be designed for importing.

Dan-in-CA, 2024-07-25:
That's really strange. I did not add that line but it was added with the commit  bc030daa2e8.
It is listed under local module imports but there in no such local module.
I have removed it.

jonathan-marsh, 2024-08-02:
Thanks, I'm not skilled enough with git to track down the source, but I hope it wasn't something I did!

closed 2024-08-02 by jonathan-marsh