Dan-in-CA/SIP #347: Low and high temperature behaviour

LuisMMMTS, 2024-09-24:
Be able to set behaviours on low and high temperatures (using a temperature sensor)
- might be stop all programs
- keep stations running continuously
- others....

Dan-in-CA, 2024-09-24:
This would be very easy to do using the node-red plugin.
That could also be a solution to your request about the fertilizer pump.

LuisMMMTS, 2024-09-26:
While it seems a good plan, having to spin-up a node-red instance for this seems overkill. Also because I think this would have a great impact for agricultural purposes, seems to me that integration with the base UI would bring much benefits. I am still not familiar with the codebase but will try to look around to see if I can manage anything