Dan-in-CA/SIP #348: Controlling the order of stations in a program

robohack, 2025-08-21:
Quite some time ago in #175 there was a request for SIP to have the ability to control the order of stations in programs.

I'm currently using OpenSprinkler to control a large system of 30 zones on our farm.  Being able to control the order that stations are run without having to rewire the controller is absolutely paramount for our use.  We have found that we sometimes need to change the zone order dynamically to allow for various tasks to be done without compromising total watering time for all zones.  Note also that during the dry season our total runtime is nearly 24 hours so reordering is the only solution to keep the water flowing 24x7.

The previous commercial controllers we used were reliable in that they always restarted their programs based on where it would be at the current time of day after a power failure, but of course they suffered from requiring rewiring to change zone order, and for a system as large as ours we also had to employ two separate controllers, juggling program start times between the controllers whenever changing zone runtimes.  The old controllers were also not remotely controllable either of course.  Not being able to change zone order meant loosing watering time for zones on days when work incompatible with watering was being done in them (e.g. mowing).

OpenSprinkler was chosen for our installation because, (a) it easily handles all zones in one controller, and (b) it has a feature that enables sorting the run order of the stations in a program by their station name, and (c) the price was right.  So this sorting feature works by simply prefixing all station names with a sequence number.  A program can then be configured to sort zones by name and it will then run each zone in the chosen order sorted by that sequence number.  It's quite clunky to set up and manage but with care and attention to detail it works reasonably well this way and thus covers the main requirement that zone order in a program be independent of wiring order.  The biggest caveat is that it only effectively handles the zone order for one program as it relies on the sequence number on the global name of each zone, but during growing months we only use one program anyway.

However OpenSprinkler has a couple of other critical limitations that its designer is unwilling to fix and this keeps me looking for a new and better solution:

- it cannot run a zone manually for testing/maintenance/etc. without first cancelling all queued zones (i.e. stopping the currently running program entirely)
  - it does have a manual zone feature, but a manual request is always added to the end of the queue even if no other zone is currently running (either all zones are currently paused, or the next queued zone is scheduled for later in the day and no zone is currently running) thus making this feature effectively useless for maintenance and testing
  - even if that queuing issue was fixed a manual request also first requires cancelling a zone if it is already queued to run later in the day (maybe this is just a UI issue as the API design could allow for it)
- it cannot restart a program mid-way through its schedule based on the current time of day
  - this makes it highly unreliable in case of power failures or user screwups -- OpenSprinkler does not and cannot recover from failures of any kind without waiting up to a full day until the next scheduled program start time
  - this also means the first problem cannot be easily worked around by cancelling a running program (cancelling all queued zones) because then one must wait up to a day until the program runs starts again

I would endeavour to modify OpenSprinkler to fix these problems myself, however it's written in C++, a language I do not get along with at all.  At least with SIP I'm not allergic to the language it is written in (though I'm no expert either).

From what I can gather SIP handles testing/maintenance easily with the "Run Once" feature -- the only caveat being that by default it stops any currently running zone for the duration of the manually run zone, but that's fine for my needs, provided the currently running program continues after the manually run zone is turned off.

However to convert to SIP I would also need for it to have some way to set the order of zones in a program.

Dan in #175 you commented that you didn't see a practical way of changing the order of zones in a program because of the way things were implemented with simple ordered lists.  However it seems to me that if programs consisted of an ordered list with members which were tuples of both the zone number and duration for that slot in the program, then it would be trivial to implement, at least on the scheduling side.  Managing the configuration in the UI might not be as trivial, but I'm sure it could be done without too much trouble.

Hopefully SIP is already as reliable as commercial controllers in the face of power failures or reboots in that it can restart a program in the middle of its runtime, however I couldn't work out whether it is or not from what I see in the documentation.

Dan-in-CA, 2025-08-22:
I did a quick test of unplugging the controller while a program was running.
When the power was restored the program did not restart. I consider this a bug and will work on fixing it.

Dan-in-CA, 2025-09-09:
finally, I pushed an update for the power cut recovery upgrade.
Testing reveled some issues that required extra work.

As of SIP revision 5.2.0 programs will finish their scheduled runs after power is restored or if SIP is otherwise restarted.

Because this update involved working on the scheduling functions I have also got some ideas about implementing a station/zone rescheduling plugin. It should be fairly easy to develop a UI page that will show a programs stations and allow them to be reordered by simply dragging and dropping them to new locations in the list.

robohack, 2025-09-09:
That's really awesome to hear Dan!

I'll hopefully get to take a much closer look once harvest wraps up.